ken-yossy/nvmetool-win #1: Mobile NVME

Fnckerpoi, 2019-11-08:
Why can't I read and recognize the mobile nvme hard disk?

ken-yossy, 2019-12-15:
What do"mobile" and "nvme hard disk" mean?
You have an HDD connected to your PC with nvme?

If you are using some protocol translation such as usb-nvme, it may not work... I have only tested against the drive directly attached to PC via PCIe.

Try to connect your drive directly to your PC via PCIe, if you can

Regards,

ken-yossy, 2020-06-30:
I will close this issue tomorrow (July 1, 2020), because it has been left for half a year.

Fnckerpoi, 2020-06-30:
> I will close this issue tomorrow (July 1, 2020), because it has been left for half a year.

Oh.thanks， I forgot this issue. 
sorry，Please close it

closed 2020-06-30 by Fnckerpoi