ken-yossy/nvmetool-win #2: Is is possible to complete "Security Receive" Admin Command?

JasonjhHuang, 2020-08-06:
Hi Ken, thanks for your great job about developing NVMe tool in Windows, it helps me a lot for understanding how to access Windows's inbox NVMe driver.
However, I'd like to know if it's possible to provide sample code about "**Security Receive**" command? I think it would be great to get more information about "level 0 discovery data" from this command. 
Please let me know how you think about this, many thanks.

ken-yossy, 2020-08-06:
Hi Jason,

Thank you for comment.

Basically, "Security Receive" command seems to be able to be issued via SCSI Pass-Through mechanism according to the Microsoft's document [1].

Also, some commands are already succeeded to be issued via SCSI Pass-Through mechanism.

So I think that it is worth to try (has possibility) to issue "Security Send" and "Security Receive" commands using Windows' inbox driver.

But I have 2 issues:
1. I'm not familiar with security protocols, so even if I could build sample code, it would be a very very simple (meaningless) code
2. I do not know "level 0 discovery data"... could you please explain it? 

[1] https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-command-set-support

JasonjhHuang, 2020-08-07:
Hi Ken,

I know "Security Receive" command is using SCSI Pass-Through and you did implement this IOCTL in nvmetool-win.

My purpose of using this command is to get some OPAL status which is defined in "Level 0 discovery data". I could get some OPAL status through ATA command 0x5C[1] on SATA disk (you could also check this function void DtaDev::discovery0() here: https://github.com/ChubbyAnt/sedutil/blob/master/Common/DtaDev.cpp).

And I thinks I could get these OPAL status through "Security Receive" on NVMe disk, or maybe I was wrong?

[1] http://www.t13.org/Documents/UploadedDocuments/docs2016/di529r14-ATAATAPI_Command_Set_-_4.pdf

ken-yossy, 2020-08-07:
Hi Jason,

> And I thinks I could get these OPAL status through "Security Receive" on NVMe disk, or maybe I was wrong?

I think you are connect. 

Because TRUSTED SEND / RECEIVE in ACS and Security Send / Receive in NVMe are just "tunnel" commands to send / receive commands and data to / from drive.

I don't have a OPAL-ready NVMe drive for testing now, and I'm afraid that I have to enable OPAL for getting "Level 0 discovery data", but using emulated NVMe drive (ex. by Qemu), I think it can be tried.

As far as I read the spec. [1], only "Security Receive" command is needed to get "Level 0 discovery data" (you already mentioned) , and other arguments and response data structure are described in [1].

If I'm wrong or misunderstanding, please let me know.

[1] Trusted Computing Group, "TCG Storage Architecture Core Specification", Version 2.01, Revision 1.00, Aug., 2015

JasonjhHuang, 2020-08-07:
Hi Ken,

It's so good to hear this, as least I didn't go wrong with this request.

I was trying to modify this function (iReadViaSCSIPassThrough) to issue "Security Receive" command and get response data from NVMe disk, however I'm not sure what parameters should be written into sptwb.

If you have any ideas please let me know, many thanks.

ken-yossy, 2020-08-07:
Hi Jason,

I think modifying iReaViaSCSIPassThrough is good way to try it.

According to the spec of SECURITY PROTOCOL IN command in SCSI (SPC-4) and the spec of Level 0 Discovery in TCG Architecture,  it might work with the following changes (not confirmed).

```
sptwb.spt.CdbLength = 12;

sptwb.spt.Cdb[0] = SCSIOP_SECURITY_PROTOCOL_IN; // A2h
sptwb.spt.Cdb[1] = 1; // CDB[1] Protocol ID
sptwb.spt.Cdb[3] = 1; // CDB[3] ComID
sptwb.spt.Cdb[4] = 0x80; // CDB[4] bit 7 : INC_512
sptwb.spt.Cdb[9] = 1; // CDB[9:6] Allocation Length; 1x512 byte
```

Even if the whole data size of "Level 0 Discovery Data" is not 512 byte, the TCG spec allows this situation that host specifies the smaller size.

I'll try it after checking the specifications and my code carefully again.
If you find something wrong, let me know.

Thank you.

ken-yossy, 2020-08-08:
I have created a branch named "test_security_recv" and committed & pushed basic code for getting Level 0 Discovery data.
The changes are almost same as the code fragment showed in the previous comment.

According to the status code, the command (SECURITY PROTOCOL IN in SCSI) seems to be succeeded, but the received data from SSD is all zero...something wrong.

According to the Command Support and Effect log, the SSD used in testing supports Security Receive command, but it might not support TCG.

I will test another SSD.

ken-yossy, 2020-08-09:
I have tested my program with a QEMU-emulated NVMe drive that I modified to declare supporting Security Receive command.
From the test, I found that the emulated SSD did not receive Security Receive command.

I think the reason why Windows returns dummy data is because Windows knows whether the system (including BIOS) supports TCG.

(Windows sometimes returns "cached" data for "Command Supported and Effects" of Get Log Page command)

I'll look for other way to test.

JasonjhHuang, 2020-08-09:
Hi Ken,

I add the following code into nvmetool-win and test it with a OPAL ready SSD.

> sptwb.spt.CdbLength = 12;
> sptwb.spt.Cdb[0] = SCSIOP_SECURITY_PROTOCOL_IN; // A2h
> sptwb.spt.Cdb[1] = 1; // CDB[1] Protocol ID
> sptwb.spt.Cdb[3] = 1; // CDB[3] ComID
> sptwb.spt.Cdb[4] = 0x80; // CDB[4] bit 7 : INC_512
> sptwb.spt.Cdb[9] = 1; // CDB[9:6] Allocation Length; 1x512 byte

And I get the response as followed which I think it's Level 0 Discovery data

>             00  01  02  03  04  05  06  07   08  09  0A  0B  0C  0D  0E  0F
>        ---------------------------------------------------------------
>  0x000  00  00  00  A0  00  00  00  01   00  00  00  00  00  00  00  00   ........ ........
>  0x010  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
>  0x020  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
>  0x030  00  01  10  0C  11  00  00  00   00  00  00  00  00  00  00  00   ........ ........
>  0x040  00  02  10  0C  09  00  00  00   00  00  00  00  00  00  00  00   ........ ........
>  0x050  00  03  10  1C  01  00  00  00   00  00  00  00  00  00  02  00   ........ ........
>  0x060  00  00  00  00  00  00  00  08   00  00  00  00  00  00  00  00   ........ ........
>  0x070  02  01  10  0C  00  00  00  09   04  00  00  00  00  00  00  00   ........ ........
>  0x080  02  02  10  0C  00  00  00  09   00  A0  00  00  00  00  00  01   ........ ........
>  0x090  02  03  10  10  07  FE  00  01   00  00  04  00  09  00  00  00   ........ ........

I'll do more test and let you know, many thanks.

ken-yossy, 2020-08-10:
Hi Jason,

It's great!! Thank you for telling me the result.

According to the TCG spec, your data can be interpreted partly as follows:

* byte [47:00] : Level 0 Discovery Header
  * byte [3:0] = 0xA0 (Length of Parameter Data, 160 byte)
  * byte [5:4] = 0x0 (Data Structure Major Version)
  * byte [7:6] = 0x1 (Data Structure Minor Version)
  * byte [15:8] = 0x0 (Reserved)
  * byte [47:16] = 0x0 (Vendor Unique)
* byte [163:48] : Feature Descriptor(s)
  * byte [49:48] = 0x0001 (Feature Code = TPer feature)
  * byte [50] = 0x10 (Version = 1)
  * byte [51] = 0x0C (Length = 12 byte)
  * byte [63:52] = (Feature Dependent Data)
  * byte [65:64] = 0x0002 (Feature Code = Locking feature)
  * byte [66] = 0x10 (Version = 1)
  * byte [67] = 0x0C (Length = 12 byte)
  * byte [79:68] = (Feature Dependent Data)

I think It definitely is a Level 0 Discovery data.

ken-yossy, 2020-08-10:
Hi Jason,

I've tested along following procedure:

1. Prepare an OPAL-ready NVMe SSD (Samsung 970 EVO Plus)
2. Set it as a data drive for Windows 10 Pro (Insider Preview)
3. Activate BitLocker for the entire drive.
4. Run nvmetool-win and try to get Level 0 Discovery data

But my code did not return valid data in the result of Security Receive command.

I filled the dataout buffer with "0x01" before issuing IoDeviceControl(), then nvmetool-win showed the pre-filled data.
So, it seems Windows does not touch the dataout buffer.

Could you please explain your environment and procedure?
Thank you.

JasonjhHuang, 2020-08-10:
Hi Ken,

My testing procedure is not that complicated.

1.  Use LITEON CA3-8D256 NVMe disk as data disk
2.  Host disk is Transcend (TS240GSSD220S) with Windows 20H1 (build number: 19041.388)
3.  Run nvmetool-win to get Level 0 Discovery data and get the response in my previous response

Why do you need to activate BitLocker? What's your Windows version?

ken-yossy, 2020-08-10:
Hi Jason,

Thank you for your explanation. It's very simple as you mentioned.

> Why do you need to activate BitLocker? What's your Windows version?

Just I thought it was necessary to activate BitLocker... Ok, I will try without activating BitLocker.
The version of Windows I used is Windows 10 Client Insider Preview (build 20175) from Dev Channel.

Thank you.

ken-yossy, 2020-08-12:
Hi Jason,

I have not succeeded in getting Level 0 Discovery data yet, but I think this depends on only my environment and  / or procedure.

As you already succeeded with above changes, I think this issue (ticket) can be marked as "resolved".
So, I propose the handling of this issue as follows:

* Close this issue after I clean related code and merge to trunk 
* I will open a new issue for tracking my further test 

If you have any comment, please let me know.

Thank you.

JasonjhHuang, 2020-08-12:
Hi Ken,

I was thinking about closing this ticket as well. Thanks for your great help and wish this tool would be kept maintained.

Thanks,
Jason

closed 2020-08-12 by JasonjhHuang

ken-yossy, 2020-08-12:
Hi Jason,

Thank you for giving me a good opportunity to add this feature.
Also your explanations were very appreciated.

Many thanks,
Ken