ken-yossy/nvmetool-win #6: telemetry can be read with IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG

tim-oleksii, 2020-09-16:
Since win10 2004 

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ni-ntddstor-ioctl_storage_get_device_internal_log

https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-feature-support

ken-yossy, 2020-09-16:
I will try it. thank you.

ken-yossy, 2020-09-17:
In [1], it says that Telemetry is available through `StorageAdapterProtocolSpecificProperty` / `StorageDeviceProtocolSpecificProperty` from `IOCTL_STORAGE_QUERY_PROPERTY`.

So I checked my code and changed to use `StorageAdapterProtocolSpecificProperty` rather than `StorageDeviceProtocolSpecificProperty` with `IOCTL_STORAGE_QUERY_PROPERTY`.

As a result, I succeeded in IoDeciveControl().

I used EXCERIA M.2 SSD by Kioxia.

Note that, this program does not support getting "Telemetry Controller/Host-Initiated Data Block"s because the SSD I'm using does not have any Controller-Initiated telemetry data and seems to be unable to create Host-Initiated telemetry data...

[1] https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-feature-support

ken-yossy, 2020-09-17:
Hi Tim,

Thank you very much for your comment.
With your comment, I found how to fix getting telemetry (header) data on Windows 10 1909.

If you are still watching this ticket, please close it if you do not mind.
Even if you do not close, I will close this ticket after a week.

Thank you
Ken

tim-oleksii, 2020-09-17:
Hi Ken,

There is a particular reason MS introduced IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG in 2004. This syscall sets `LSP` bit. However, data layout is quite different, it doesn't comply to NVME spec, but it's quite close to it.
Additionally, this one is fast - data is read with regards to MDTS.

Both approaches are feasible.

I've checked all the points with modified FW on my eSSD.

Best regards,
Alexey

ken-yossy, 2020-09-17:
Hi Alexey,

Thank you for your further information.

> There is a particular reason MS introduced IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG in 2004. This syscall sets `LSP` bit. However, data layout is quite different, it doesn't comply to NVME spec, but it's quite close to it.
> Additionally, this one is fast - data is read with regards to MDTS.

It's interesting. I'm just thinking about how to create telemetry log.

> Both approaches are feasible.
> 
> I've checked all the points with modified FW on my eSSD.

It's great. Ok, I'll try using `IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG` to create and get Host-Initiated Telemetry Log Data with Windows 10 2004 (from Windows Insider Preview).

Regards,
Ken

ken-yossy, 2020-09-17:
I have created branch named "issue-6".
I will use this branch for this issue.

tim-oleksii, 2020-09-17:
There is one subtle issue with this syscall. It's output applies `little-endianess` on WRD (16 bits).
There is no way to get size of resutling telemetry from the header, as ioctl regenerates it every time by setting `LSP`.

So the only solution - allocating 32 MBs at once (max telemetry size) and reading the whole log page

ken-yossy, 2020-09-20:
merge some commits to this branch from master branch for issue #7

ken-yossy, 2020-10-02:
I just wrote as [1] says and tested with an NVMe SSD with Telemetry support, but got error (error code = 31 "A device attached to the system is not functioning") from `DeviceIoControl()`. There is something wrong in my code.

Note that issuing `Get Log Page` command via `DeviceIoControl()` with `IOCTL_STORAGE_QUERY_PROPERTY` to the same device is succeeded.

```
    GET_DEVICE_INTERNAL_STATUS_DATA_REQUEST request;
    PDEVICE_INTERNAL_STATUS_DATA outBuffer = NULL;
    ULONG outBufferLength = 0;

    // Allocate buffer for use.
    outBufferLength = sizeof(DEVICE_INTERNAL_STATUS_DATA) + 512;
    outBuffer = malloc(outBufferLength);
    if (outBuffer == NULL)
    {
        vPrintSystemError(GetLastError(), "malloc");
        goto error_exit;
    }

    ZeroMemory(&request, sizeof(GET_DEVICE_INTERNAL_STATUS_DATA_REQUEST));
    ZeroMemory(outBuffer, outBufferLength);

    // from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ns-ntddstor-get_device_internal_status_data_request
    request.Version = sizeof(GET_DEVICE_INTERNAL_STATUS_DATA_REQUEST);
    request.Size = sizeof(GET_DEVICE_INTERNAL_STATUS_DATA_REQUEST);
    request.RequestDataType = DeviceCurrentInternalStatusDataHeader;

    // from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ns-ntddstor-device_internal_status_data
    outBuffer->Version = sizeof(DEVICE_INTERNAL_STATUS_DATA);
    outBuffer->StatusDataLength = 512;
    outBuffer->Size = outBufferLength + 512;

    // Send request down.
    iResult = iIssueDeviceIoControl(_hDevice,
        IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG,
        &request,
        sizeof(GET_DEVICE_INTERNAL_STATUS_DATA_REQUEST),
        outBuffer,
        outBufferLength,
        &returnedLength,
        NULL
    );
```
[1] https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ni-ntddstor-ioctl_storage_get_device_internal_log

tim-oleksii, 2020-10-03:
@ken-yossy  Do you need code sample?

ken-yossy, 2020-10-03:
> @ken-yossy Do you need code sample?

Yes, if you don't mind. I appreciate it.
Now, I'm building Qemu based environment to capture commands and arguments from Windows to SSD.

Ken

ken-yossy, 2020-10-04:
I've built the environment with Windows 10 Insider Preview build 20175, and tested the tool with above code.

The result was that Windows actually issued the Get Log Page command (with Create Telemetry Host-Initiated Data = 1) and the emulated SSD model received it.

I think this result means there's something wrong along response path.

tim-oleksii, 2020-10-05:
This one was checked against our eSSD

``
#include <Windows.h>
#include <iostream>
#include <ntddstor.h>
#include <string>
#include <vector>
#include <sstream>
#include <fstream>

namespace
{
	std::string dev_path = "";
	auto request_data_type = DeviceCurrentInternalStatusDataHeader;
	auto request_data_set = DeviceStatusDataSetUndefined;
	size_t size_to_read = 512;
	std::string output_file_name = "";
};

std::string dt_to_str(DEVICE_INTERNAL_STATUS_DATA_REQUEST_TYPE dt)
{
	switch (dt)
	{
		case DeviceInternalStatusDataRequestTypeUndefined:
			return "DeviceInternalStatusDataRequestTypeUndefined";
		case DeviceCurrentInternalStatusDataHeader:
			return "DeviceCurrentInternalStatusDataHeader";
		case DeviceCurrentInternalStatusData:
			return "DeviceCurrentInternalStatusData";
		default:
			return "Unknown(" + std::to_string(static_cast<unsigned>(dt)) + ")";
	}
}

std::string ds_to_str(DEVICE_INTERNAL_STATUS_DATA_SET ds)
{
	switch (ds)
	{
	case DeviceStatusDataSetUndefined:
		return "DeviceStatusDataSetUndefined";
	case DeviceStatusDataSet1:
		return "DeviceStatusDataSet1";
	case DeviceStatusDataSet2:
		return "DeviceStatusDataSet2";
	case DeviceStatusDataSet3:
		return "DeviceStatusDataSet3";
	case DeviceStatusDataSet4:
		return "DeviceStatusDataSet4";
	case DeviceStatusDataSetMax:
		return "DeviceStatusDataSetMax";
	default:
		return "Unknown(" + std::to_string(static_cast<unsigned>(ds)) + ")";
	}
}

void print_help()
{
	std::cout << 
"Args:\n\
   -d - device path, mandatory.\n\
   -dt - RequestDataType, integer in range [0, 2]\n\
   -ds - RequestDataSet, integer in range [0, 5]\n\
   -s - size to read, default 512\n\
   -o - output file name to store internal log\n\
Example:\n\
   telemetry_read -d  \\\\.\\PhysicalDrive0 -dt 1 -ds 2\n\
   telemetry_read -d  \\\\.\\PhysicalDrive0 -dt 1 -ds 2 -s 4096 -o c:\\tmp\\telemetry.bin";
}

[... 168 lines cut ...]
                             IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG,
                             &request,
                             sizeof(request),
                     &out_buffer.front(), out_buffer.size(), &bytes_returned, nullptr);

    if (ioctl_result == 0)
	{
		auto last_error = GetLastErrorAsString();
		std::cout << "Failed\n";
		std::cout << "Error: " << last_error << "\n" ;
		return -1;
	}
	else
	{
		std::cout << "Succeeded\n";
	}

	CloseHandle(dev);

	std::cout << "BytesReturned = " << bytes_returned << "\n";
	print_status_data(*reinterpret_cast<DEVICE_INTERNAL_STATUS_DATA*>( & out_buffer.front()));

	return 0;
}
``

tim-oleksii, 2020-10-05:
@ken-yossy you may find snippet here
https://pastebin.pl/view/416a9efb

ken-yossy, 2020-10-06:
@tim-oleksii 
Thank you very much for your sample.
With it, I have succeeded in DeviceIoControl().
I finally understood your comments in this ticket about the data layout (NVMe spec and DEVICE_INTERNAL_STATUS_DATA) and endianess...

Currently, I've not understood why it works and the data (DEVICE_INTERNAL_STATUS_DATA) received from (emulated) SSD does not seem as I expects.

I will do some tries to resolve them.

Ken

tim-oleksii, 2020-10-06:
@ken-yossy  sorry, I can't help with QEMU. eSSD was used to grab telemetry data. Actually, this is the first time I hear about SSD emulation. Are there any insights how to do it?

ken-yossy, 2020-10-07:
@tim-oleksii don't worry. This is a matter regarding to the way I select.

> Actually, this is the first time I hear about SSD emulation. Are there any insights how to do it?

Qemu has emulation codes of NVMe SSD(s) and can provide emulated NVMe SSD to guest OSes. So, I install Windows 10 Insider Preview as a guest OS on qemu, edit the emulation codes, expose it to the Windows, and test this tool with changing SSD's behavior.

I think it's almost the same as your way; the difference is that you are using real SSD and changing real FW.

One thing that I have found along this way is that installing recent Windows 10 (including Insider Preview) as a guest OS on qemu is very difficult. I encountered errors many times and restarted installation from the beginning each time...so, I strongly recommend using real devices if possible :-)

I will push edited codes of this tool and summarize the behavior of Windows as far as I understood along this trial and error.

Thank you,
Ken

ken-yossy, 2020-10-07:
I created [wiki page](https://github.com/ken-yossy/nvmetool-win/wiki/Getting-telemetry-data-with-IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG) for this topic and wrote observations from the results of experiments. I'm sorry for rough description...

ken-yossy, 2020-10-08:
I've merged the changes made in the branch "issue-6" into main (master) branch.

ken-yossy, 2020-10-16:
Along this ticket, it is confirmed that we can create Host-Initiated Telemetry (by setting "Create Telemetry Host-Initiated Data" bit) and retrieve the telemetry header with IOCTL_STORAGE_GET_DEVICE_INTERNAL_LOG using Windows' inbox NVMe device driver.

But, the NVMe spec says the content of Telemetry data is vendor specific, and I don't know  real SSDs that create and export Telemetry data.

So I would like to close this ticket a week later.

If I find such SSDs and any issue in this tool,  I will open a new issue at that time.

Thank you,
Ken

closed 2020-10-24 by ken-yossy