ken-yossy/nvmetool-win #7: C++ support

tim-oleksii, 2020-09-17:
Is there any reason why this project considered as C++ one? I see `#include <iostream>` or `#include <string>` but they don't seem to be used.

ken-yossy, 2020-09-18:
Hi Alexey,

Thank you for your comment and reminding me of that.

> Is there any reason why this project considered as C++ one? I see `#include <iostream>` or `#include <string>` but they don't seem to be used.

No, because I've thought it as low priority, I have simply left it in that state so far.
I will to make it right.

Ken

ken-yossy, 2020-09-18:
I plan to put this issue ahead of #6.

ken-yossy, 2020-09-20:
With the recent sequence of commits, this project has been recognized as pure C project.

@tim-oleksii 
If you are still watching this ticket, do you mind closing it?
Even if you do not close, I will close it after a week.

Again, thank you for your comments.

Ken

closed 2020-09-20 by tim-oleksii