ken-yossy/nvmetool-win #8: no firmware download/commit

tim-oleksii, 2020-10-22:
I see that this application lacks FW processing routines. Is it possible to add them?

One trick - since Win10 2004 only CA=3 is supported.

ken-yossy, 2020-10-23:
Hi Alexey,

Thank you for your comment.

As you mentioned, this tool does not support Firmware Update (image download and commit) feature defined in NVMe spec.
But it is intended because FW update is highly dangerous process and also drive (manufacture) dependent.
For the worst case, SSDs could turn into bricks...

So I am not going to add them at this time.

Regards,
Ken

ken-yossy, 2020-10-31:
If there's no further comment, I would like to close this ticket a week later. 

Thank you,
Ken

closed 2020-11-08 by ken-yossy