ken-yossy/nvmetool-win #10: [build] PDEVICE_INTERNAL_STATUS_DATA not define

djukyo, 2021-02-02:
VS2019 build failed.
PDEVICE_INTERNAL_STATUS_DATA not define.
Could you help check it?

ken-yossy, 2021-02-03:
Hi @djukyo,
Thank you for your comment.

> PDEVICE_INTERNAL_STATUS_DATA not define.
> Could you help check it?

Could you please check the version of Windows Driver Kit (WDK) you're using?
I'm using WDK version 10.0.18346.1000, and VS2019 finds PDEVICE_INTERNAL_STATUS_DATA in winioctl.h in my environment.

djukyo, 2021-02-10:
You are right.
Build it need VS2019 + WDK.
Today I installed WDK(19041.685.201201-2105.vb_release_svc_prod1_WindowsSDK).

djukyo, 2021-02-10:
Add more:
D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe --version
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe --help
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe list
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

djukyo, 2021-03-01:
@ken-yossy could you help me check it?

ken-yossy, 2021-03-01:
> @ken-yossy could you help me check it?

Hi @djukyo, about what do you want me to check? 
Are you still in trouble with undefined symbol (PDEVICE_INTERNAL_STATUS_DATA)?

djukyo, 2021-03-02:
how to run?
D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe --version
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe --help
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

D:\4_Repo\3_Public\nvmetool-win\x64\Debug>NVMeTool.exe list
[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified.

ken-yossy, 2021-03-02:
> how to run?

This tool does not accept options and sub-commands that you specified ('--version', '--help', and 'list'), simply because they are not implemented.

djukyo, 2021-03-02:
Thanks for your feedback.
Which features have been implemented?
I want to try run it.

ken-yossy, 2021-03-02:
> Which features have been implemented?
> I want to try run it.

This tool provides accesses to NVMe drives (typically SSDs) via Windows inbox device driver, for example, getting drive's fundamental data and statistical data, reading data from drive.
Please carefully read README file for further information, before using this tool.
Thank you.

ken-yossy, 2021-03-04:
If you have any other questions, please issue individual ticket.
Thank you.

closed 2021-03-04 by ken-yossy