ken-yossy/nvmetool-win #12: Windows on ARM, Support?

sidd-kishan, 2021-05-30:
Please let us know when can we have an ARM64 version for Windows on ARM OS.

ken-yossy, 2021-05-31:
Hi @sidd-kishan,

At this time, I'm not working on testing this tool on Windows10 on ARM, because I don't have a hardware...
Surface Pro X is pretty expensive, and Lenovo IdeaPad 4G seems to be also expensive (over $1,000) for testing this tool.

Recently I heard that Qualcomm's Snapdragon Developer Kit (w/ Windows10 on ARM) will be available in this summer.
If it can use NVMe drives and is reasonable, I'll buy it and test this tool on it.

Regards,
Ken

sidd-kishan, 2021-05-31:
We at Windows on Rasberry Pi community have ported windows on arm to Rasberry pi 4 b+ we can help you with the testing. It is WOA deployment and supports ARM64 Drivers. we would be glad to lend you a hand in testing the drivers you build.

ken-yossy, 2021-05-31:
Thank you for your proposal, but honestly speaking, I'm not in a hurry for testing and porting this tool for Windows on ARM.
 
Well... would you mind if you test [this script](https://github.com/ken-yossy/nvmetool-win-powershell) on WOA?
It's a simple PowerShell script that also access NVMe drive using Windows' inbox device driver like this tool.
If it works, I think this tool will probably work on WOA.
Any feedback is appreciate.

Regards,
Ken

closed 2021-07-07 by stale[bot]