ken-yossy/nvmetool-win #13: ProtocolDataRequestSubValue4 support

tim-oleksii, 2021-06-16:
Hi Ken,

MS claims to support setting LSP, RAE bits via ProtocolDataRequestSubValue4. I guess you can reflect this in your tool to read telemetry.

ken-yossy, 2021-06-17:
Hi Alexey,

Thank you for your information.
As you say, I found following description in _[Working with NVMe drives](https://docs.microsoft.com/en-us/windows/win32/fileio/working-with-nvme-devices#protocol-specific-queries)_.

```c++
protocolData->ProtocolDataRequestSubValue4 = 0; // This will map to STORAGE_PROTOCOL_DATA_SUBVALUE_GET_LOG_PAGE definition, then user can pass Retain Asynchronous Event, Log Specific Field.
```

I will try reading Host-Initiated Telemetry... and fix some bugs :-)

Regards,
Ken

tim-oleksii, 2021-06-21:
Hi @ken-yossy ,

Yeah, you got it. Testing it on 21h1 has yielded no results, I suspect that 21h1 doesn't support it.
I haven't tried Windows 11 yet.

Best regards,
Alexey

ken-yossy, 2021-06-21:
Thank you for your information.
I will keep your test result in mind.
Also, I will make a test code and continue testing with each release of Windows Insider Preview.

Regards,
Ken

ken-yossy, 2021-07-16:
Changed codes to use LSP and RAE bits via ProtocolDataRequestSubValue4 at issuing Get Log Page command with retrieving Host Initiated Telemetry Data.
But the behavior does not change...as Alexey said.

tim-oleksii, 2021-08-15:
As far as I understand, version of Windows 10 >22000 is able to gather telemetry

ken-yossy, 2021-08-15:
Thank you for your information. I will test on Windows OS build no. over 22000 (Windows 11?) with its ISO image (it's convenient).
So, I will close this issue shortly, and will reopen when the ISO image is released.

Thank you,
Ken

closed 2021-08-15 by ken-yossy