ken-yossy/nvmetool-win #14: Issuing Format NVM  Fail

CharlesJuGit, 2021-06-23:
Hi Ken-Yossy,
During usage of the Tool, 
I tried the commands and found that I could not issue Format NVM.
And the error message is: 
[E] DeviceIOControl : (ErrorCode=1) Incorrect function.
[E] Failed in issuing Format NVM command by DeviceIOControl (IOCTL_STORAGE_REINITIALIZE_MEDIA)

I'm wondering to know if there is some setting or config I should modify or change?
Thanks for your kindly help.

Environments as below:
MB  :  Gigabyte X570 UD
CPU :  AMD Ryzen 3 3100 4-Core Processor
OS:  Windows 10 10.0.10586

Here is the attachment picture.
[image]

ken-yossy, 2021-06-23:
Hi Charles,

Thank you for your comment.
I will investigate it.

Could you please tell me the build number of Windows 10 of your system?
You can get it in "System" > "About" as follows.
[image]

Kind regards,
Ken

CharlesJuGit, 2021-06-24:
Hi  Ken,
Here is the update:
Test Result 1: 
[image]
OSInfo: Windows 10 10.0.10586 (It is 1511 threshold 2 by wikipedia https://ja.wikipedia.org/wiki/Microsoft_Windows_10%E3%81%AE%E3%83%90%E3%83%BC%E3%82%B8%E3%83%A7%E3%83%B3%E5%B1%A5%E6%AD%B4)

Test Result 2:
[image]
OSInfo: 19042.1052(It is 20H2 by wikipedia https://ja.wikipedia.org/wiki/Microsoft_Windows_10%E3%81%AE%E3%83%90%E3%83%BC%E3%82%B8%E3%83%A7%E3%83%B3%E5%B1%A5%E6%AD%B4)
[image]

The Test Result2  environment is 20H2 version (19042.1052) with Visual Studio 2019

Thanks for your reply.

Best Regards,
Charles.

eeric7777, 2021-06-24:
Hello,

I thought the IOCTL_STORAGE_REINITIALIZE_MEDIA API is only enabled in WinPE mode ?  
or it's just my misunderstanding ? 

https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-command-set-support

ken-yossy, 2021-06-24:
Hi @eeric7777,

It says:

> Only enabled in Win PE mode for IOCTL_STORAGE_PROTOCOL_COMMAND. SCSIOP_SANITIZE for IOCTL_SCSI_PASS_THROUGH. IOCTL_STORAGE_REINITIALIZE_MEDIA only supports crypto erase.

I think this means the following three independent limitations.

* Issuing `Format NVM` by `IOCTL_STORAGE_PROTOCOL_COMMAND` is limited to WinPE mode
* To issue `Format NVM` by `SCSIOP_SANITIZE`, `IOCTL_SCSI_PASS_THROUGH` should be used
* `IOCTL_STORAGE_REINITIALIZE_MEDIA` only supports "Cryptographic Erase" in NVMe spec; Secure Erase Settings (SES) should be `0b010`

Regards,
Ken

ken-yossy, 2021-06-27:
Hi all,

I tried `FormatNVM` command with Samsung 970 EVO Plus on Windows 10 Insider Preview (build 21354), and succeeded as follows.

So this phenomenon probably depends on drive and/or version (build) of Windows.
Next, I will try to issue the command with the drive on Windows 21H1 (build 19043).
Also, I will try to find drives other than the above one that `FormatNVM` command can be issued with this tool among drives that I have.

```
PS C:\users\ken\downloads> .\NVMeTool.exe 1
[I] nvmetool-win version 2.0

[I] Running on Windows 10 build 21354
[I] Getting controller identify data succeeded.

[M] SerialNumber (SN): -----
[M] Model Number (MN): Samsung SSD 970 EVO Plus 500GB
[M] Firmware Revision (FR): 2B2QEXM7
[M] Version (VER): 0x00010300 (NVMe Revision 1.3.0)

[I] Getting controller's SMART / Health information log page succeeded.

# Input command:
#  - r: Read data from LBA = 0
#  - w: Write data to LBA = 0
#  - d: Deallocate LBA = 0
#  - l: Get Log Page
#  - i: Identify
#  - g: Get Feature
#  - s: Set Feature
#  - f: Flush
#  - t: Device Self-test
#  - z: Format NVM
#  - v: Security Receive
#
#  press 'q' to quit program

> i

# Input Controller or Namespace Structure (CNS) value (in hex):
#    Supported CNS values are:
#     00h = Namespace data structure (NSID = 1)
#     01h = Controller data structure (CNTID = 0)
#     02h = Active Namespace ID list
#     03h = Namespace Identification Descriptor (NSID = 1)

> 1

# Identify : Controller data structure (CNTID = 0), Press 'y' to continue

> y

(snip)

[M] Format NVM Attributes (FNA):
        bit [      2] 1 = Supports cryptographic erase operation
        bit [      1] 0 = Cryptographic erase or user data erase as part of a format is performed on a per namespace basis.
        bit [      0] 1 = Format operation applies to all namespaces

(snip)

# Input command:
#  - r: Read data from LBA = 0
#  - w: Write data to LBA = 0
#  - d: Deallocate LBA = 0
#  - l: Get Log Page
#  - i: Identify
#  - g: Get Feature
#  - s: Set Feature
#  - f: Flush
#  - t: Device Self-test
#  - z: Format NVM
#  - v: Security Receive
#
#  press 'q' to quit program

> z

# Format NVM : CAUTION!! all data are erased on the target drive
# Format NVM : Press 'y' to continue

> y

[I] Issuing Format NVM command by DeviceIoControl(IOCTL_STORAGE_REINITIALIZE_MEDIA) is succeeded
```

Regards,
Ken

ken-yossy, 2021-06-27:
> Next, I will try to issue the command with the drive on Windows 21H1 (build 19043).

I succeeded on Windows 10 21H1 (build 19043) with Samsung 970 EVO Plus.

My next action is the following. 
> Also, I will try to find drives other than the above one that FormatNVM command can be issued with this tool among drives that I have.

And I will check what arguments are passed from Windows to drives using qemu.

ken-yossy, 2021-06-27:
Hi Charles,

Could you tell me the "Format NVM Attributes (FNA)" of your drive?
You can get it with `Identify` command with CNS = 1 (Controller Data Structure).

You will see something like that:
```
[M] Format NVM Attributes (FNA):
        bit [      2] 1 = Supports cryptographic erase operation
        bit [      1] 0 = Cryptographic erase or user data erase as part of a format is performed on a per namespace basis.
        bit [      0] 1 = Format operation applies to all namespaces
```

I think one possible reason is that some arguments along `FormatNVM` command your drive expects may differ from Windows passes. 

The attribute is one of indications of it.

Thank you,
Ken

CharlesJuGit, 2021-06-27:
Hi Ken,
The FNA is got by identify command with CNS=1 below:
[image]
[image]
[image]
I will also research if there is some value different with standard case.
And I will try to use another device for testing by Tuesday.
Thank you.
Best Regards,
Charles

ken-yossy, 2021-06-27:
Hi Charles,

Thank you for your information.
It's interesting that bit 0 of "Format NVM Attributes" is different from Samsung 970 EVO Plus.
I will check NVMe spec to find what behavior is expected with the value of this field, too.

CharlesJuGit, 2021-06-28:
Hi Ken,
I also found a possible factor which is [O]Sanitize Capabilities(SANICAP).
The device shows:
[O]Sanitize Capabilities (SANICAP):
bit[2]0=Does not support Overwrite sanitize operation
bit[1]1=Supports Block Erase sanitize operation
bit[0]1=Supports Crypto Erase sanitize operation

Maybe this is another factor of "Format operation is specific to a particular namespace".

Thank you,
Best Regards,
Charles

ken-yossy, 2021-06-29:
I captured the arguments of `FormatNVM` command that Windows 10 Insider Preview (build 21354) issued (by `IOCTL_STORAGE_REINITIALIZE_MEDIA`)

Then I found that Windows issued `FormatNVM` command with the following arguments regardless of the value of FNA bit 0.
For CDW10, bit 10 means "Cryptographic Erase", and bit 4 can be ignored (about metadata). 

```
[NVME] >>> Format NVM
[NVME] >>>	 NSID  = 0x00000001
[NVME] >>>	 PRP1  = 0x0000000000000000
[NVME] >>>	 PRP2  = 0x0000000000000000
[NVME] >>>	 CDW10 = 0x00000410
[NVME] >>>	 CDW11 = 0x00000000
[NVME] >>>	 CDW12 = 0x00000000
[NVME] >>>	 CDW13 = 0x00000000
[NVME] >>>	 CDW13 = 0x00000000
[NVME] >>>	 CDW13 = 0x00000000
```

I will check NVMe spec for `FormatNVM` command that specified an NSID.

Another point of view:
In my test, all target NVMe drives were not mounted ("offline" and volumes were not created on them).
Therefore, I thought that Windows might not have issued the `FormatNVM` command because the target drive (volume) was mounted.

I tested....but Windows issued the command to a drive even if it was online and the volume on it was mounted.

Thank you,
Ken

ken-yossy, 2021-07-03:
According to the NVMe spec (1.3d), for `Format NVM` command with Secure Erase (`SES` field is not zero), controller changes its behavior by bit 0 of `FNA` field in Identify Controller data structure.

If bit 0 of `FNA` field is zero, the format operation is affected to the namespace that the Namespace ID (NSID) of the command specifies; NSID = 0xFFFFFFFFh is not allowed.
If bit 0 of `FNA` field is one, the format operation is affected to the all namespace in the NVM subsystem (SSD) as long as the NSID is valid one or 0xFFFFFFFh.

[image]

Devices we tested support cryptographic erase for Secure Erase, and Windows seems to issue `Format NVM` command with requesting cryptographic erase.

As described above, for `Format NVM` command with cryptographic erase, the bit 0 of `FNA` is referred for the behavior of the drive, but we found that Windows specified 0x1 for NSID. It seemed to be a valid NSID.
 
So, at this time, the situation we examined here seems to depend on the drive.

Thank you,
Ken

CharlesJuGit, 2021-07-05:
Hi Ken,
In your investigation it seems to be related with FNA bit 0 and NSID.
I will check this part and drive.

Thank you,
Best Regards,
Charles.

CharlesJuGit, 2021-07-16:
Hi Ken,
So far I'm still investigating it.
And I would like to try out other method like sanitize or secure erase.
I'll close this issue first.
Thanks for your kindly help.

Best Regards,
Charles.

closed 2021-07-16 by CharlesJuGit