ken-yossy/nvmetool-win #15: How to use IOCTL_STORAGE_FIRMWARE_DOWNLOAD?

NextStory1127, 2021-07-28:
Update the Windows device through IOCTL_STORAGE_FIRMWARE_DOWNLOAD, the windows computer directly prompts the crash prompt "SYSTEM_SERVICE_EXCEPTION", How to use IOCTL_STORAGE_FIRMWARE_DOWNLOAD?

thanks

ken-yossy, 2021-07-28:
Hi,

I think the usage of IOCTL_STORAGE_FIRMWARE_DOWNLOAD is well described in [this page](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/upgrading-firmware-for-an-nvme-device).
But I have not confirmed that the code really works... because updating firmware is highly dangerous (this is the reason that I do not implement it).

Regards,
Ken

NextStory1127, 2021-07-30:
Thank you very much, successfully upgraded the firmware through IOCTL_SCSI_MINIPORT

ken-yossy, 2021-07-31:
Great! I'm glad to hear that and it's my pleasure.

Regards,
Ken

closed 2021-08-02 by NextStory1127