ken-yossy/nvmetool-win #18: Example command to run identify controller to get SUBNQN field

paulmenzel, 2021-08-19:
The Linux kernel prints an error if the SUBNQN field is invalid or missing. This can be verified with `nvme id-ctrl /dev/nvme0n1` on Linux with [nvme-cli](https://github.com/linux-nvme/nvme-cli/).

Unfortunately, the Dell support asks for a command to verify this under Microsoft Windows, which I do not have access to. The [example outputs in your Wiki](https://github.com/ken-yossy/nvmetool-win/wiki) contain the SUBNQN field. Can you please tell me how to get those example outputs with your tool, so I can forward it to the Dell support?

ken-yossy, 2021-08-19:
Hi,

I think if you try to forward this tool to them you need to build this tool because only source files are maintained in this repository.
But I'm afraid that forwarding pre-built binary ("*.exe") often causes trouble... because environments are generally different between you and them (also between you and me).

So I have made Powershell script (`scripts\get-subnqn.ps1` in the repository https://github.com/ken-yossy/nvmetool-win-powershell) that retrieve SUBNQN in the Identify Controller data from an NVMe drive using Windows' inbox driver.

You can retrieve SUBNQN and some other fields related to target drive as follows.
The argument is the physical drive number of the target.

```
PS > ./get-subnqn.ps1 0
Serial Number (SN): ************
Model Number (MN): WDC PC SN720 SDAQNTW-512G-1001
Firmware Revision (FR): 10190101
NVM Subsystem NVMe Qualified Name (SUBNQN): nqn.2018-01.com.wdc:nguid:************-0001-001B448B448DF0E5
```
(Note: serial number is changed to asterisk `*`)

With only an Windows 10 system that NVMe drive and Powershell script works, they can retrieve it from the drive with this script.

I hope it help your situation.

Kind regards,
Ken

paulmenzel, 2021-08-20:
Thank you so much for your reply, which exceeds all my expectations. Fingers crossed, that Dell and SK hynix are going to [fix the firmware of the SK hynix BC511](https://lore.kernel.org/linux-nvme/20210817170222.GA224912@dhcp-10-100-145-180.wdc.com/T/#rf4b78390d7f7b09684faf2087a7a6e9a7e6eaa58).

closed 2021-08-20 by paulmenzel