ken-yossy/nvmetool-win #19: Compiler error

KerickHuang, 2021-08-21:
VS2019 build this project can't find PDEVICE_INTERNAL_STATUS_DATA struct in #include <ntddstor.h>@NVMeGetTelemetry.c

ken-yossy, 2021-08-21:
Hi,

Thank you for your information.
But it can be built without errors in my environment. 
Would you please check that Windows Driver Kit is installed in your environment (10.0.19030.1000 in my environment)?

[image]

Kind regards,
Ken

KerickHuang, 2021-08-21:
You got it thank you, it's solved and I can build project pass now.

I am a beginner, want to write and read LBA to SSD on Windows

Can this tool do it?

Command line argument how to set?
set 2 C:\ but it not find dev following
[image: image.png]

ken-yossy, 2021-08-22:
Hi,

> You got it thank you, it's solved and I can build project pass now.

I'm glad to hear that!

> I am a beginner, want to write and read LBA to SSD on Windows Can this tool do it?

No, currently starting LBA and number of sectors are fixed, starting LBA is 0 and number of sectors is 1 (hard coded).

> Command line argument how to set?

It does not have any command line arguments related to the drive's action, such as starting LBA, number of sectors, and the data source / destination.
It only takes the physical drive number of the target NVMe drive.

The starting LBA and number of sectors to be read or written are specified in NVMeSCSIPassThrough.c
The below is for read operation.
```c
int iReadViaSCSIPassThrough(HANDLE _hDevice)
{
(snip)

    if ( srbType )
    {
        ZeroMemory(&sptwb_ex, sizeof(SCSI_PASS_THROUGH_WITH_BUFFERS_EX));
(snip)
        sptwb_ex.spt.DataInTransferLength = 512;
(snip)
        sptwb_ex.spt.Cdb[0] = SCSIOP_READ;
        sptwb_ex.spt.Cdb[5] = 0; // Starting LBA
        sptwb_ex.spt.Cdb[8] = 1; // TRANSFER LENGTH
(snip)
    }
    else
    {
        ZeroMemory(&sptwb, sizeof(SCSI_PASS_THROUGH_WITH_BUFFERS));
(snip)
        sptwb.spt.DataTransferLength = 512;
(snip)
        sptwb.spt.Cdb[0] = SCSIOP_READ;
        sptwb.spt.Cdb[5] = 0; // Starting LBA
        sptwb.spt.Cdb[8] = 1; // TRANSFER LENGTH

(snip)
    }
    return iResult;
}
```
Number of bytes to be read (multiple of sector size) is set to `sptwb_ex.spt.DataInTransferLength` and `sptwb.spt.DataTransferLength`.

Also, Number of sectors to be read is set to `sptwb_ex.spt.Cdb[8]` and `sptwb.spt.Cdb[8]`.

Starting LBA to be read is set to `sptwb_ex.spt.Cdb[5]` and `sptwb.spt.Cdb[5]'.

Note that structure `SCSI_PASS_THROUGH_WITH_BUFFERS` and `SCSI_PASS_THROUGH_WITH_BUFFERS_EX` have buffer read data goes to inside them.
If you want to read more than one sector or to use a drive of non-512 byte sector (ex. 4kB sector), you need to adjust these parameters and buffer.

You can modify this tool on your own, to accept arbitrary starting LBA and number of sectors, and files that read data goes to / write data exists.

`CDB` (Command Data Block) is a data format specified in SCSI Block Commands (SBC).
If you need further information about CDB, you should check the specification.

I hope it help you.

Thank you,
Ken

KerickHuang, 2021-08-22:
Hi Ken,

Thank you for your advice.

> Hi,
>
>
> I'm glad to hear that!
>
>
> No, currently starting LBA and number of sectors are fixed, starting LBA is 0 and number of sectors is 1 (hard coded).
>
>
> It does not have any command line arguments [...]

ken-yossy, 2021-08-23:
I think the original problem (the title of this issue) has been resolved.
I close this ticket.

Thank you,
Ken

closed 2021-08-23 by ken-yossy