ken-yossy/nvmetool-win #21: Error while running NVMeTool.exe

ankitsingh1508, 2021-09-16:
Bit new to this, so might be doing some silly mistake.
Have downloaded and compiled nvmetool-win as per your readme and was able to successfully generate NVMETool.exe

But while trying to run (NVMeTool.exe PhysicalDrive0) it am getting below error:-
"[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified."

Not sure if something is wrong with my setup., can you please help me in resolving this issue.?

ken-yossy, 2021-09-16:
Hi,

Thank you for your comments.

> But while trying to run (NVMeTool.exe PhysicalDrive0) it am getting below error:-
> "[I] Running on Windows 10 build 18363[E] CreateFile: (error code = 2) The system cannot find the file specified."
> 
> Not sure if something is wrong with my setup., can you please help me in resolving this issue.?

It needs only the digit of the physical drive name to be accessed.
In your system, it may work by `NVMeTool.exe 0` (`0` from `PhysicalDrive0`).

If you have any further trouble, feel free to tell me.

I will add some examples for running it into README.

Kind regards,
Ken

ankitsingh1508, 2021-09-17:
Hi Ken,

Thanks a lot for the response.
Its working with "NVMeTool.exe 0" and able to fetch details, though it failed for "DeviceIoControl"

YES, it would be a great help if you add few examples in README.
Am looking for firmware update command, basically intention is to write some non firmware file to boot partition.

Regards,
Ankit Singh

ken-yossy, 2021-09-17:
Hi Ankit,

> Thanks a lot for the response.
> Its working with "NVMeTool.exe 0" and able to fetch details, though it failed for "DeviceIoControl"

I'm glad to hear that!

> Am looking for firmware update command
> "Firmware Commit", // 10h
> "Firmware Image Download", // 11h
> Basically intention is to write some non firmware file to boot partition.

As I [commented in an another issue](https://github.com/ken-yossy/nvmetool-win/issues/15#issuecomment-887990453), the procedure for updating firmware of NVMe drive is not implemented to this tool but is well described in Microsoft's web page.

I think the description would help you.

Regards,
Ken

ankitsingh1508, 2021-09-17:
Hi Ken,

Thanks for guiding, will look into Microsoft's web page.
Can you please give me example command for "Get Log Page"

Regards,
Ankit Singh

ken-yossy, 2021-09-17:
Hi Ankit,

> Can you please give me example command for "Get Log Page"

Well...`Get Log Page` command is the entry point to get various information / statistics / status of NVMe drive, for example, SMART information, Firmware slot information.

To specify the information to be retrieved, Log Identifier is used (2 for SMART information, 3 for Firmware Slot Information). 

In this tool, the procedure for getting SMART information is implemented in `iNVMeGetSMART()` in NVMeSMART.c, and the one for getting Firmware Slot Information is implemented in `iNVMeGetFwSlotInformation()` NVMeFwSlotInformation.c (most of these codes are interpretation of the response ><).

For `Get Log Page` command, [Microsoft's web page](https://docs.microsoft.com/en-us/windows/desktop/fileio/working-with-nvme-devices) is also helpful.

Regards,
Ken

ankitsingh1508, 2021-09-17:
Hi Ken,

On a linux based nvmi-cli utility am using "get-log" to read DATA from boot partition.
For example :- 
sudo nvme get-log /dev/nvme0n1 --log-id=0x15 --lpo=16 --log-len=16384 --lsp=0 -b >> sample_1.txt

So was just confused how to call similar command on this NVMeTool.exe..?

Regards,
Ankit Singh

ken-yossy, 2021-09-17:
> On a linux based nvmi-cli utility am using "get-log" to read DATA from boot partition.
> For example :-
> sudo nvme get-log /dev/nvme0n1 --log-id=0x15 --lpo=16 --log-len=16384 --lsp=0 -b >> sample_1.txt
> 
> So was just confused how to call similar command on this NVMeTool.exe..?

I've just got what you want to do!

Unfortunately, this tool does not support "nvme-cli"-like non-interactive mode nor dumping response data into file.
Only interactive, and console-printing mode is available.

I think that implementing the two features based on this tool is not very difficult.
You can implement features only you need, and change the behavior as you like.

Regards,
Ken

ankitsingh1508, 2021-09-17:
Sure got it, will try and implement it.

Currently for call to "DeviceIoControl" from WinFunc.c is failing.
Is it due to me using stornvme.sys version 10.0.19041.1202 instead of recommended version of 10.0.21354.1 ?

Regards,
Ankit Singh

ken-yossy, 2021-09-17:
Hi Ankit,

> Currently for call to "DeviceIoControl" from WinFunc.c is failing.
> Is it due to me using stornvme.sys version 10.0.19041.1202 instead of recommended version of 10.0.21354.1 ?

No, I don't think so (depends on what you are trying).
Along my experience, checking arguments for DeviceIoControl() might resolve the situation.

But please note that what we can / cannot do using Windows' inbox device driver is NOT clearly disclosed.

In [this page](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-feature-support), Microsoft partially describes supported functions / features for NVMe drives using Windows' inbox driver.
We need to make trials-and-errors....

Regards,
Ken

ankitsingh1508, 2021-09-20:
Hi Ken,

Thanks a lot.
Able to do firmware update using "IOCTL_SCSI_MINIPORT"

Regards,
Ankit Singh

ken-yossy, 2021-09-20:
It's great!

So, could you please close this issue?

Thank you,
Ken

ankitsingh1508, 2021-09-21:
Sure, will close it.

Before that 1 last query... :-)
NVMe firmware update is usually done by copying the firmware file to Boot Partition and then then activate it, which is working fine in this case.
But if i try to copy a non firmware file to boot partition - its failing might be due to some expected file format.
Can we skip this verification and copy any non-firmware file to boot partition?

Regards,
Ankit Singh

ken-yossy, 2021-09-21:
Hi Ankit,

> Before that 1 last query... :-)

No problem :-)

> NVMe firmware update is usually done by copying the firmware file to Boot Partition and then then activate it, which is working fine in this case.

Yes, exactly.

> But if i try to copy a non firmware file to boot partition - its failing might be due to some expected file format.
> Can we skip this verification and copy any non-firmware file to boot partition?

I'm afraid that it's difficult... the function (procedure) of verifying firmware image is completely hidden to us (users), and more over, the method of verification may depend on vendor's choice.
Also, I've not found the "switch" to turn on/off the verification in NVMe spec.
Even if there's a drive that has "switch" that can be set from host side via NVMe command, it would be implemented as vendor specific command / setting.

If you know the verification method for specified device, you could make the file data to pretend to be a proper firmware for the device.

Note: I assume the term "Boot Partition" means "Firmware Slot" (hidden storage for firmware), not head sector (sector its LBA is zero). 

Regards,
Ken

ankitsingh1508, 2021-09-21:
Thank for the clarification.

Linix nvme-cli fw-download and fw-commit commands gives this option based on action value:
		action=1 – image activated after controller level reset, or 
		action=3 – image activated immediately    
		action=6 – for writing to boot partitions
So was thinking if something similar can be done here.

Regards,
Ankit Singh

closed 2021-09-21 by ankitsingh1508