ken-yossy/nvmetool-win #22: Can you add a function for sending a vendor-specific command?

kobyou, 2021-11-03:
Can you add a function for sending a vendor-specific command to NVME SSD? thanks!

ken-yossy, 2021-11-04:
Hi!

> Can you add a function for sending a vendor-specific command to NVME SSD? thanks!

Well, I don't really want to implement the feature into this tool, because I cannot test the feature with real drives (as the name says, the specification of "vendor-specific commands" are not disclosed to users!)

As much as possible, I could add sample and non-working code into it (needs to modify to work).
 
If you have any specific vendor command(s) for your drive(s) now, you can implement the function according to the [Microsoft's article](https://docs.microsoft.com/en-us/windows/win32/fileio/working-with-nvme-devices#example-sending-a-vendor-specific-command).

Source code of this tool will help you to make the partial code in the Microsoft's article working.

Kind regards,
Ken

ken-yossy, 2021-11-05:
Hi!

I've just added a sample (template) code to issue vendor specific commands to NVMe drives.
You can find staffs in `NVMeAdminVendorSpecificCommand.c`.
You can change the opcode, parameters, length of return value, and so on, as you like (including adding input from file(s) and output to file(s)).

I have confirmed with QEMU emulated NVMe drive that Windows' driver passed all the arguments (CDW10 to CDW15) to the drive as programmed.

Thank you,
Ken

closed 2021-12-12 by stale[bot]