ken-yossy/nvmetool-win #24: Error while running NVMeTool.exe

freeqimmy, 2022-01-30:
Hi Ken,
First of all, thank you so much for making this amazing tool, it's very helpful for a NVMe beginner like me.
But on the first time I try to run NVMeTool.exe it suggests that there is an I/O device error. 
The error message is:
[I] Running on Windows 11 build 22000[E] DeviceIoControl: (error code = 1117) The request could not be performed because of an I/O device error.
[E] Getting controller identify data failed, stop.

Environments are as below:
CPU: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   2.80 GHz
OS:  Windows 11 22000.469
Dependencies: Visual Studio Community 16.8.4 + Windows Driver Kit 10.0.20348.0

Here is my attachment photo:
[image]

Looking forward to your reply, that will be much of help!

Regard,
freeqimmy

ken-yossy, 2022-01-31:
Hi @freeqimmy,

Thank you for your information.

Well... could you please check the drive '0' is really NVMe drive directly attached to the system or not.
Because this tool cannot handle NVMe drives connected via protocol conversion like USB-NVMe enclosures.

If your drive is directly attached to the system, please check the drive number (id) '0' is correct.
You can check it with "Disk Management" tool (you can see it in the menu showed by right click on Windows' logo), like following:

[image]

Kind regards,
Ken

freeqimmy, 2022-01-31:
Hi Ken,
I checked Disk Management page and it looks like this:
[image]

Seems the disk 0 does exist, but somehow it cannot be recognized.

Would this problem be related to the driver which my OS is using? It seems my laptop runs storage driver provided by intel (as shown in the pic below), and i see intel rapid storage technology turned on in BIOS (and unfortunately i cannot turn it off). Maybe if i am using Standard NVM Express Controller Driver this problem would not be happened, but i don't know for sure.
[image]

Thanks for your reply, and may you have a happy tiger year!

Regards,
freeqimmy

ken-yossy, 2022-01-31:
Hi @freeqimmy 

I hope you're also having a happy tiger year!
And also thank you for your further information.

> Would this problem be related to the driver which my OS is using? It seems my laptop runs storage driver provided by intel (as shown in the pic below), and i see intel rapid storage technology turned on in BIOS (and unfortunately i cannot turn it off). 

I think your assumption is almost correct!

Because there are some reports (below is Lenovo's document, as example) that Intel VMD (Volume Management Device) 'hides' PCIe NVMe drives:
_[NVMe SSD PCI device will be invisible when Intel VMD is enabled - Lenovo ThinkSystem
](https://support.lenovo.com/us/en/solutions/ht506197-nvme-ssd-pci-device-will-be-invisible-when-intel-vmd-is-enabled-lenovo-thinksystem)_

When a system is using Microsoft's inbox NVMe device driver, Device Manager shows the configuration as below (viewed with "devices by connection"):

[image]

I'm afraid that you see "Intel RST VMD (Managed) Controller 9A0B" instead of "Standard NVM Express Controller" in your system.

I have found articles that some mother board vendors will provide new BIOS that makes changeable the configuration of Intel VMD and/or disable it by default.

So I recommend to check the web site of the mother board vendor or manufacture of your system if you can.

I hope it is helpful for you.

Regards,
Ken

ken-yossy, 2022-01-31:
I forgot one important thing for this tool.
This tool only supports Microsoft's inbox NVMe device driver.

Thank you,
Ken

closed 2022-03-10 by stale[bot]