ken-yossy/nvmetool-win #25: Wrong value in ProtocolDataRequestSubValue for NVMeDataTypeLogPage

Calises, 2022-05-09:
According to M$ doc https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ne-ntddstor-_storage_protocol_nvme_data_type :
for [IOCTL_STORAGE_QUERY_PROPERTY](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ni-ntddstor-ioctl_storage_query_property) with `NVMeDataTypeLogPage`, field `ProtocolDataRequestSubValue` in [STORAGE_PROTOCOL_SPECIFIC_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ns-ntddstor-_storage_protocol_specific_data) is offset for log page, not NamespaceID. Insert `NVME_NAMESPACE_ALL` as log page offset will cause Error 1117 :)
I noticed this problem for `NVME_LOG_PAGE_HEALTH_INFO` and `NVME_LOG_PAGE_COMMAND_EFFECTS`.

ken-yossy, 2022-05-10:
Hi Calises,

Thank you for your comment! I have found the description you mentioned.
I will fix them and test.

Regards,
Ken

Calises, 2022-05-10:
Great, thx!

closed 2022-06-11 by ken-yossy