ken-yossy/nvmetool-win #26: write, read fail

fadu-dufresne, 2022-05-23:
Hello!

I tried to read(-r), write(-w) to my nvme device using the tool you produced, but it failed. Scsi status comes out as 02h.
But 'get log page' works very well.
Here is snapshot.

- read
[image]
- write
[image]
- get log page (02h)
[image]

Can you tell me what the problem is?

fadu-dufresne, 2022-05-27:
Thank you for your help.

I checked my CDB option, but it has no problem, I think. T-T

[image]

I am also using Windows 10, so there seems to be no environmental difference.

I will keep looking for the cause. Thank you again. :)

ken-yossy, 2022-06-01:
Hi!

I've just found that Microsoft says that SCSI commands `READ(16)` and `WRITE(16)` are used to issue Read and Write commands to NVMe drive via SCSI PassThrough mechanism in [this page](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-command-set-support), but I use `READ(10)` and `WRITE(10)` in this tool.

So, this difference might be one of the causes of your problem.

Could you please try applying the following patch to `NVMeSCSIPassThrough.c` and sending `Read` command to your NVMe drive?
With this patch, this tool changes to use `READ(16)` and `WRITE(16)` instead of `READ(10)` and `WRITE(10)`.

In my environment, after applying this patch, `Read` command for an NVMe SSD has successfully completed.

```
@ -50,6 +50,7 @@ typedef struct _SCSI_PASS_THROUGH_DIRECT_WITH_BUFFER_EX
 #define CDB6GENERIC_LENGTH                   6
 #define CDB10GENERIC_LENGTH                  10
 #define CDB12GENERIC_LENGTH                  12
+#define CDB16GENERIC_LENGTH                  16
 #define SETBITON                             1
 #define SETBITOFF                            0

@ -586,7 +587,7 @@ int iWriteViaSCSIPassThrough(HANDLE _hDevice)
         sptdwb_ex.sptd.Version                  = 0;
         sptdwb_ex.sptd.Length                   = sizeof(SCSI_PASS_THROUGH_DIRECT_EX);
         sptdwb_ex.sptd.ScsiStatus               = 0;
-        sptdwb_ex.sptd.CdbLength                = CDB10GENERIC_LENGTH;
+        sptdwb_ex.sptd.CdbLength                = CDB16GENERIC_LENGTH;
         sptdwb_ex.sptd.StorAddressLength        = sizeof(STOR_ADDR_BTL8);
         sptdwb_ex.sptd.SenseInfoLength          = SPT_SENSE_LENGTH;
         sptdwb_ex.sptd.DataOutBuffer            = (PVOID)databuffer;
@ -602,9 +603,9 @@ int iWriteViaSCSIPassThrough(HANDLE _hDevice)
         sptdwb_ex.StorAddress.Target            = 0;
         sptdwb_ex.StorAddress.Lun               = 0;
         sptdwb_ex.sptd.SenseInfoOffset          = offsetof(SCSI_PASS_THROUGH_DIRECT_WITH_BUFFER_EX, ucSenseBuf);
-        sptdwb_ex.sptd.Cdb[0]                   = SCSIOP_WRITE;
-        sptdwb_ex.sptd.Cdb[5]                   = 0; // Starting LBA
-        sptdwb_ex.sptd.Cdb[8]                   = 1; // TRANSFER LENGTH
+        sptdwb_ex.sptd.Cdb[0]                   = SCSIOP_WRITE16;
+        sptdwb_ex.sptd.Cdb[9]                   = 0; // Starting LBA
+        sptdwb_ex.sptd.Cdb[13]                  = 1; // TRANSFER LENGTH

         length = sizeof(SCSI_PASS_THROUGH_DIRECT_WITH_BUFFER_EX);
         iResult = iIssueDeviceIoControl(
@ -642,16 +643,16 @@ int iWriteViaSCSIPassThrough(HANDLE _hDevice)
         sptdwb.sptd.PathId              = 0;
         sptdwb.sptd.TargetId            = 0;
         sptdwb.sptd.Lun                 = 0;
-        sptdwb.sptd.CdbLength           = CDB10GENERIC_LENGTH;
+        sptdwb.sptd.CdbLength           = CDB16GENERIC_LENGTH;
         sptdwb.sptd.SenseInfoLength     = SPT_SENSE_LENGTH;
         sptdwb.sptd.SenseInfoOffset     = offsetof(SCSI_PASS_THROUGH_DIRECT_WITH_BUFFER, ucSenseBuf);
         sptdwb.sptd.DataIn              = SCSI_IOCTL_DATA_OUT;
         sptdwb.sptd.DataTransferLength  = sectorSize;
         sptdwb.sptd.TimeOutValue        = 5;
         sptdwb.sptd.DataBuffer          = databuffer;
-        sptdwb.sptd.Cdb[0]              = SCSIOP_WRITE;
-        sptdwb.sptd.Cdb[5]              = 0; // Starting LBA
-        sptdwb.sptd.Cdb[8]              = 1; // TRANSFER LENGTH
+        sptdwb.sptd.Cdb[0]              = SCSIOP_WRITE16; // 8Ah
+        sptdwb.sptd.Cdb[9]              = 0; // Starting LBA
+        sptdwb.sptd.Cdb[13]             = 1; // TRANSFER LENGTH

         length = sizeof(SCSI_PASS_THROUGH_DIRECT_WITH_BUFFER);
         iResult = iIssueDeviceIoControl(
@ -687,7 +688,7 @@ int iReadViaSCSIPassThrough(HANDLE _hDevice)
         sptwb_ex.spt.Version = 0;
         sptwb_ex.spt.Length = sizeof(SCSI_PASS_THROUGH_EX);
         sptwb_ex.spt.ScsiStatus = 0;
-        sptwb_ex.spt.CdbLength = CDB10GENERIC_LENGTH;
+        sptwb_ex.spt.CdbLength = CDB16GENERIC_LENGTH;
         sptwb_ex.spt.StorAddressLength = sizeof(STOR_ADDR_BTL8);
[... 15 lines cut ...]
@ -728,16 +729,16 @@ int iReadViaSCSIPassThrough(HANDLE _hDevice)
         sptwb.spt.PathId = 0;
         sptwb.spt.TargetId = 0;
         sptwb.spt.Lun = 0;
-        sptwb.spt.CdbLength = CDB10GENERIC_LENGTH;
+        sptwb.spt.CdbLength = CDB16GENERIC_LENGTH;
         sptwb.spt.SenseInfoLength = SPT_SENSE_LENGTH;
         sptwb.spt.SenseInfoOffset = offsetof(SCSI_PASS_THROUGH_WITH_BUFFERS, ucSenseBuf);
         sptwb.spt.DataIn = SCSI_IOCTL_DATA_IN;
         sptwb.spt.DataTransferLength = 512;
         sptwb.spt.TimeOutValue = 5;
         sptwb.spt.DataBufferOffset = offsetof(SCSI_PASS_THROUGH_WITH_BUFFERS, ucDataBuf);
-        sptwb.spt.Cdb[0] = SCSIOP_READ;
-        sptwb.spt.Cdb[5] = 0; // Starting LBA
-        sptwb.spt.Cdb[8] = 1; // TRANSFER LENGTH
+        sptwb.spt.Cdb[0] = SCSIOP_READ16; // 88h
+        sptwb.spt.Cdb[9] = 0; // Starting LBA
+        sptwb.spt.Cdb[13] = 1; // TRANSFER LENGTH

         length = offsetof(SCSI_PASS_THROUGH_WITH_BUFFERS, ucDataBuf) + sptwb.spt.DataTransferLength;
         iResult = iIssueDeviceIoControl(
```

Regards,
Ken

closed 2022-07-10 by stale[bot]