ken-yossy/nvmetool-win #27: For reading NVMe - how to change Offset from default (0) to required (say 1024) ?

loke786, 2022-09-16:
Hi,

Am able to read from NVMe Ok, but, with mention params tried changing Offset to configure as reading from particular point (say 12) , but receiving data from beginning part only instead specified Offset, any thoughts about param to change or suggestions would be really appreciated. 
..
protocolData->ProtocolDataRequestValue = NVME_LOG_PAGE_..;
protocolData->ProtocolDataRequestSubValue = 12; // lower 32-bit of the offset
protocolData->ProtocolDataRequestSubValue2 = 12; // higher 32-bit of the offset
..

Thanks,
Lokesh

ken-yossy, 2022-09-16:
Hi,

I think that you are correct that using `ProtocolDataRequestSubValue` and `ProtocolDataRequestSubValue2` to specify the offset of the retrieving data. The following is excerpt from `winioctl.h`. 

```
typedef enum _STORAGE_PROTOCOL_NVME_DATA_TYPE {
(snip)

    NVMeDataTypeLogPage,        // Retrieved by command - GET LOG PAGE
                                // Corresponding values in STORAGE_PROTOCOL_SPECIFIC_DATA,
                                //      ProtocolDataRequestValue - Log page id
                                //      ProtocolDataRequestSubValue - Lower 32-bit offset value
                                //      ProtocolDataRequestSubValue2 - Upper 32-bit offset value
                                //      ProtocolDataRequestSubValue3 - Log specific identifier
```

So If you want to retrieve log data from offset 12 (0xC), I think you can get expected data as follows.
(SubValue2 should be zero)

```
protocolData->ProtocolDataRequestSubValue = 12; // 0xC / lower 32-bit of the offset
protocolData->ProtocolDataRequestSubValue2 = 0; // higher 32-bit of the offset
```

Note that `LOPL` and `LPOU` have been added into the spec from NVMe revision 1.3.
Please check the NVMe revision of your device.

I will check the behavior with SMART log data.

Hope it help you.

Thank you,
Ken

loke786, 2022-09-16:
Hello Ken, Thanks for the quick response.

I tried with mention change and seen same behavior (as reading from beginning) with setup NVMe 1.3 - Win11.

protocolData->ProtocolDataRequestSubValue = 12; // 0xC / lower 32-bit of the offset
protocolData->ProtocolDataRequestSubValue2 = 0; // higher 32-bit of the offset

Thank you,
Lokesh

ken-yossy, 2022-09-16:
Hi,

Thank you for prompt testing and providing further information!
It is helpful for me to check my code.

Regard,
Ken

ken-yossy, 2022-09-19:
Hi Lokesh,

At first, I say sorry to you because your drive must support "extended data for the Get Log Page command" feature in NVMe spec to use LOPL and LPOU.

You can check it with the Identify Controller data of your drive.
Bit 2 of "Log Page Attribute" field shows that the drive supports LOPL/LPOU or not.
Following is the Identify Controller data of my drive (This drive supports it).

```
[M] Log Page Attributes (LPA):
        bit [      3] 1 = Supports Telemetry Host-Initiated and Telemetry Controller-Initiated log pages and sending Telemetry Log Notices
        bit [      2] 1 = Supports extended data for Get Log Page (including extended Number of Dwords and Log Page Offset fields)
        bit [      1] 1 = Supports Commands Supported and Effects log page
        bit [      0] 0 = Does not support SMART / Health information log page on a per namespace basis
```
And with this drive, I have confirmed that getting log page data with expected offset by setting to `protocolData->ProtocolDataRequestSubValue` is working.

The following data dump is SMART data getting with no offset.
You can see 0x6D at byte offset 0x60.
```
        00  01  02  03  04  05  06  07   08  09  0A  0B  0C  0D  0E  0F
        ---------------------------------------------------------------
 0x000  00  40  01  64  05  00  00  00   00  00  00  00  00  00  00  00   .@.d.... ........
 0x010  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x020  0C  BF  13  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x030  34  CE  14  00  00  00  00  00   00  00  00  00  00  00  00  00   4....... ........
 0x040  EB  CA  76  01  00  00  00  00   00  00  00  00  00  00  00  00   ..v..... ........
 0x050  6E  EB  5F  01  00  00  00  00   00  00  00  00  00  00  00  00   n._..... ........
 0x060  6D  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   m....... ........
 0x070  46  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   F....... ........
 0x080  57  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   W....... ........
 0x090  03  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x0A0  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x0B0  DD  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x0C0  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
```

And the following is one with offset 0x20 (32 bytes):
You can see 0x6D at byte offset **0x40** (0x40 = 0x60 - 0x20), so specified offset seems to be working.
```
        00  01  02  03  04  05  06  07   08  09  0A  0B  0C  0D  0E  0F
        ---------------------------------------------------------------
 0x000  FC  BE  13  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x010  19  CE  14  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x020  E0  C9  76  01  00  00  00  00   00  00  00  00  00  00  00  00   ..v..... ........
 0x030  E8  E8  5F  01  00  00  00  00   00  00  00  00  00  00  00  00   .._..... ........
 0x040  6D  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   m....... ........
 0x050  46  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   F....... ........
 0x060  57  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   W....... ........
 0x070  03  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x080  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x090  DD  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
 0x0A0  00  00  00  00  00  00  00  00   00  00  00  00  00  00  00  00   ........ ........
```

Hope it help you.

Thank you,
Ken

loke786, 2022-09-19:
Hello Ken, Thanks for the detailed help.

I found like on AHCI mode results as expected per ProtocolDataRequestSubValue, but not on RAID, so, might be some other parameter / approach (NVME_IOCTL_PASS_THROUGH) needed, will check. Any example / thoughts on same would be really helpful please, thanks!

ken-yossy, 2022-09-20:
Unfortunately, I don't have information about what and how commands can be issued via SCSI Passthrough mechanism other than simple Read, Write and Trim (deallocate).

I'm afraid that it will get more complicated if RAID controller and/or bridge chip is in the middle of CPU and NVMe drive(s)...

Regards,
Ken

loke786, 2022-09-20:
Sure, Ken.

From net found (below link) commands structure references... and while checking using facing DeviceIoControl 1117 error :( 
https://github.com/intel/dptf/blob/master/ESIF/Products/ESIF_CM/Sources/esif_nvme.h

In searching RAID mode sample example or documentation/references, not much luck so far. 
Share to me if any available please. Thanks in advance!

eeric7777, 2022-09-20:
If your device is controlled by hardware RAID controller or software RAID controller, you need to use its API.
However, not all APIs are open source.

For example, check this for Intel RST RAID:
https://github.com/hiyohiyo/CrystalDiskInfo/blob/a68d8592b65133248de547a5284ebc898e23e57a/AtaSmart.cpp

loke786, 2022-09-20:
Sure, Ken.

Thanks a lot for your prompt help!

closed 2022-09-20 by loke786